christianhaitian/arkos #103: Feature request: add Prince of Pérsia to Ports

heavydevi, 2020-12-20:
Arkos is by far the neste One on rg351p.
For me is just missing the clássic Prince of Pérsia from Ports, like other os on This system.
For me it would bem awesome to include it.
Thanks!

SRGilbert, 2020-12-23:
> Arkos is by far the best One on rg351p.
> For me is just missing the clássic Prince of Pérsia from Ports, like other os on This system.
> For me it would bem awesome to include it.
> Thanks!

What version though?  In my opinion the very best version of the original is the SNES port, although the Genesis port is fun too if less ambitious.

christianhaitian, 2020-12-23:
SDLPoP (Opensource Prince of Persia) has been added with today's update.  Just Update from the options menu/section.

closed 2020-12-23 by christianhaitian

heavydevi, 2020-12-23:
This is great, thanks man!

xEden95x, 2023-06-01:
@christianhaitian I'm not sure if I' missing something here, but I can't see SDLPoP on my RG353V after updating to the latest version. Would be great to be able to play it!

christianhaitian, 2023-06-01:
I don't know of retroarch port sdlpop.  I believe the stock os can play portmaster games if you load them manually.  The portmaster interface may be problematic on the stock os.

xEden95x, 2023-06-01:
The issue I'm having is that if I play it through the stock OS, it doesn't track playtime of Portmaster games. Is there any way around that?

christianhaitian, 2023-06-01:
I don't know for Stock OS.  I don't use that.