christianhaitian/arkos #105: Feature Request: Add Jazz Jackrabbit, Duke Nukem 3D and Shadow Warrior Ports to ArkOS

Gordon1972, 2020-12-22:
Are there any ports of these games available that can be easyly integrated into ArkOS?
Are there any plans?

Thanks for reading.

BTW: Thanks to christianhaitian for fixing bugs so quickly! Keep up the good work on ArkOS!

christianhaitian, 2020-12-22:
There’s good folks on discord who tried porting duke nukem 3d and had no success yet.  Jazz jackrabbit maybe possible on dosbox pure.  Don’t know about shadow warrior yet..

Gordon1972, 2020-12-22:
Thanks for the Info. The DOS-Version of Jazz Jackrabbit however is not optional. The DOSBox-Version used in ArkOS
cannot enable the Gate A20 functionality...

christianhaitian, 2020-12-22:
I'm referring to dosbox-pure.  It's not in ArkOS yet.  Maybe part of this evening's update.

SRGilbert, 2020-12-23:
Duke 3D would be awesome, I’m having a blast with lzdoom, it’s just perfection.  Only thing I haven’t figured out was how to pan and zoom on the map.

christianhaitian, 2020-12-23:
Dosbox_pure is included with today's update.  Just Update from the options menu/section.  Let me know how Jazz Jackrabbit works on this if at all.

Gordon1972, 2020-12-23:
Thanks for adding DOSBox Pure. Jazz Jackrabbit runs fine with it (a few graphical glitches due to it's special framerate) so i think we wouldn't need a standalone Jazz Port immediatly - i'm fine with it running under DOSBox Pure.

Duke Nukem 3D runs, too albeit it's a little bit slow (would say 10-15fps?) but it works inkl. music.

Also thanks for implementing SDLPoP and Half-Life, Prince of Persia works like a charm. I'll test Half-Life later.

Merry X-Mas to all who report bugs here, only to make ArkOS better. Big hugs go out to christianhaitian for maintaining ArkOS and for implementing things so fast! We really really apprechiate that!

christianhaitian, 2021-01-03:
Eduke32 and shadow warrior will be included as part of ports install menu I’m currently working on adding.  Look for uodates on this in the near future.

closed 2021-01-03 by christianhaitian