christianhaitian/arkos #107: Feature request: Add ports for GZdoom and Spelunky Classic

danielthewise, 2020-12-23:
Amazing work with arkOS. It's by far my favorite. I may just be dreaming at this point, but I would love to have a port of Spelunky Classic running on the 351p. Is this even remotely possible? I'm pretty sure the classic version of the game is open source. Someone made an android port of it that works on the Retroid Pocket 2.

I'd also love to have a port of GZdoom so I can play some awesome mods like Chex Quest 3. Is there a snowball's chance in hell of either of these? Thanks for considering!

christianhaitian, 2020-12-23:
I'm not sure on this one just yet.  I will research this further.

danielthewise, 2020-12-27:
Thanks for considering it. The following links may be helpful for spelunky.
https://github.com/yancharkin/SpelunkyClassicHD
https://www.moddb.com/games/spelunky-classic/downloads/spelunky-source-code

christianhaitian, 2021-01-03:
GZdoom requires more capable hardware than what the rg351p has to run.  I will review the possibility of porting spelunky classic

danielthewise, 2021-01-07:
Yeah, since I submitted this request, I have since realized that LZdoom is actually sufficient to play the mods I wanted to play. Thanks for the great work on that!

Spelunky Classic would be absolutely incredible. Thanks again for looking into it.

christianhaitian, 2021-01-10:
I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

closed 2021-01-10 by christianhaitian