christianhaitian/arkos #108: Downloadable ArkOS Themes

beatlesdbp, 2020-12-23:
Hello, Is there anyway you can add a feature where we can download additional EmulationStation Themes much like the Batocera software build? Thanks!

Disfunktional, 2021-01-02:
@christianhaitian I think this would be a great addition... Although with the 'ArkOS bar' it means that themes people have spec'd for rg351p won't display correctly and aspects become offset... Would it be possible to include turning that bar on/off with this request?

Otherwise the benefits of having the theme downloader reduced somewhat.

christianhaitian, 2021-01-02:
I don't intend to provide this feature at this time.  More development time that I don't have to spare on this.  Anyone wanting to make changes to the interface can do so [here](https://github.com/christianhaitian/EmulationStation-fcamod) and replace the existing emulationstation in ArkOS with it.

christianhaitian, 2021-01-27:
After further review and consideration, I don't plan to implement a downloadable themes option.  With the themes being readily available from the roms folder, it's simple enough for anyone to add them as the so choose from a PC or Mac similar to how roms can be added.

closed 2021-01-27 by christianhaitian