christianhaitian/arkos #110: Feature request: Add ports for 2048

Deer-Cr, 2020-12-24:
A port game named 2048 I think is so good. I tried to place 2048.sh file of ports in 351elec to my arkOS ports, but it can't work. What can I do to fix it? Or can you add it to arkOS?Many thanks for your work. Below is the 2048.sh file I copy from 351elec.
[2048.sh.txt](https://github.com/christianhaitian/arkos/files/5738375/2048.sh.txt)

christianhaitian, 2020-12-24:
I’ll need to find the source code first then compile it for ArkOS.

Deer-Cr, 2020-12-25:
Many thanks again

christianhaitian, 2020-12-30:
This will be included as a selectable port that you can install in the next upcoming update.  Stay tuned.

Deer-Cr, 2021-01-01:
Awesome! I'm very excited about this. Thanks

christianhaitian, 2021-01-03:
This has been delayed due to an issue.  Working on addressing it as quickly as possible.

Deer-Cr, 2021-01-05:
That's all right, still thanks for your efforts!

christianhaitian, 2021-01-29:
This has been accommodated with this evening's update.  Just update from the options section.

closed 2021-01-29 by christianhaitian