christianhaitian/arkos #111: Volume level is always reading 100%

crazyjoe29, 2020-12-24:
The volume reads at 100% no matter if you have it all the way to zero, plz fix this. Because it would nice to know what volume it's at. Also if you can add the volume over clock feature which was done by holding Right joystick down pressing up or down on the d-pad depending on how loud you wanted it. You could do all this with the stock emuelec os  the rg351p came with and it would be a bit of a godsend to be able to do this again, plz try to put this feature on the next arkOS update , I'm very sure that many people who use your amazing OS would love this feature too. Plz see what you can do, thanks

christianhaitian, 2020-12-24:
Volume will read 100 percent because it doesn’t adjust with the volume wheel because it’s not possible. The volume indicator is there because other similar devices like the odroid go adavance and rgb10 need button combinations to control volume as there’s no volume wheel.  Why add volume control using a key combination for the rg351p when we have a physical volume wheel on the rg351p?

crazyjoe29, 2020-12-24:
I think you misunderstood my askance about the volume control, the button combo  was used to make the player louder and I liked the way it could go up to 296% over the normal 100% volume level when I had been using the old os it came with when I bought the rg351p. I think it was a very good feature  to have especially for people who are not able to hear very well and it would enhance their experience too for this is a great OS and I just wanted to have everyone to have a great experience with it as much  as I do.

christianhaitian, 2020-12-24:
So you mean for retroarch based emulation.  That’s the only program that allows this as far as I know.  You can set this yourself using the retroarch menu settings.  While in game, press select + X or in just retroarch, go to hot keys and assigned your preferred keys to volume Up and volume Down.  This will allow to set your volume gains using select and what ever keys you set.  You can even change your hotkey to something else besides select if you want.

crazyjoe29, 2020-12-24:
Thanks for your help with the problem but I would like to see if you can make it easier to have the volume gains without too many steps because this makes it harder to use than if you'd  create a few easier ways to do this please let me know if you can make it possible and again thanks for everything you do for us.

christianhaitian, 2020-12-24:
To make it possible as a global update, i would have to assign it as select+up and select+down.  Retroarch doesn’t support more than 1 assigned hotkey for something like this.  So this combination would be awkward for gaming unless I change the hotkey to something else besides select which then means people would have to learn a new hotkey.  I don’t know if this is something that’s worthwhile at the moment as it seems more of a personal preference.  I’ll leave this issue open for a few days for input from others on this.

crazyjoe29, 2020-12-24:
Thank you for your consideration of this idea, I had thought it was a good idea for people who are not as well versed in this OS and tech savvy with this kind of thing to have a hot key that would easier to use and which can be able to be used when a ROM is not being played on it so they can set it up to the volume level they need or can handle. Some people who are in a noisy area might need to make it louder than normal to hear the game they are playing and a quick and easy  to remember and use hot key may be very helpful to some people because your hot keys that are being used now kinda aren't that  well known and all the way easy to use. Not that the way you have it now isn't any good, it's just that most people who use this OS don't know them enough to be a problem with the global population who use this wonderful OS. I hope you are not insulted by my asking to see if you can get them changed and if all the hot keys new and old  be posted somewhere easy to see  so people know what they are and do or not do. Please and thank you, you are an amazing coder to be able to do a OS like this and I wish you have a very happy holiday and a happy new year.

christianhaitian, 2020-12-24:
We'll see if anyone else has an opinion on this.  For noisy environment, I just use headphones so I can hear my game and not contribute to the noisy environment that others may not care for.

SRGilbert, 2020-12-27:
I would actually like an option to remove the volume indicator on ArkOS since it’s basically superfluous on the RG351P.

pentaenix, 2021-01-01:
> 
> 
> I would actually like an option to remove the volume indicator on ArkOS since it’s basically superfluous on the RG351P.

same

christianhaitian, 2021-01-01:
As this distribution is also used on other rk3326 units that don't have a volume wheel, this would fragment the management of this OS as it would add one more component I'd have to manage across different devices.  If you're skill enough, you can create your own modified emulationstation-fcamod by forking my modified source code for it [here](https://github.com/christianhaitian/EmulationStation-fcamod.git)

christianhaitian, 2021-01-03:
I'm closing this request as there's been no additional feedback from others on if this is a worthwhile feature to add.  At this time, I recommend just adding this configuration yourself through retroarch.  It may be a few steps, but once configured, you won't have to touch it again.

closed 2021-01-03 by christianhaitian