christianhaitian/arkos #112: User equalizer needed.

crazyjoe29, 2020-12-24:
To enhance the sound of the game music and add control of the sound quality of the player's speakers for a better gaming experience for the user of this OS would be much appreciated, thanks

christianhaitian, 2020-12-24:
This is not a high end device with high end speakers that will accommodate a feature like this.

closed 2020-12-24 by christianhaitian

crazyjoe29, 2020-12-24:
then why is it so expensive to buy and why is so great at emulating so many
systems and have so many features many handhelds don't have and why is it
so powerful for the price. I think it can handle it, don't you, especially
since a feature  like background music was added where you can add your own
music to the player and play it during the game as well as the PSP and ds
need ways to enhance their soundtracks and it would help your users gaming
experience become even better than it already is. Also headphones can be
connected to the device, so to say it wouldn't useful for the speaker's
sound quality is silly to say in the least because the custom equalizer
would be able to be used with headphones not just the systems speakers and
make the sound quality of the system better and control of that would help
the speakers with future updates for this OS and help the usability and
control of the handhelds speaker's sound would help with bass blurging and
allow the speakers life expectancy to be longer than they would be
normally. So plz do tell me, how this would not be useful to anyone else?
I'd like to know, so please answer me when you are able to, thanks.

christianhaitian, 2020-12-24:
It’s a limitation of the device.  Yes it can do many things but it was not designed with high end components like a nintendo switch or sony psp or vita.

crazyjoe29, 2020-12-24:
yes, I have already known that but it's still a possibility because even
cheaper handheld's like the mp5 and the powkiddy's x16 and up can do this,
I know this because I own both and the both of them are not in the same
league as the rg351p, so maybe tweaking the settings and the background
music player for the BG music folder might have some amazing results and
take a few looks at the the handhelds as mentioned above and in how they do
this and achieve this same ability  and a look at their coding for sound
equalizers might help you code the same abilities into the arkOS for the
rg351p.