christianhaitian/arkos #113: Issues with FTP

OriginalNotmatthew, 2020-12-24:
I’ve updated ArkOS multiple times, I have a compatible dongle, I cannot SSH into my device IP, cannot FTP via filezila, and cannot connect in browser (or on mobile device) to http://rg351p. I have enabled/disabled connections multiple times and wifi is working fine for multiple scraper types.
**ALL THAT SAID**
I just want to do something simple. Unfortunately the theme I use is missing logos, most notably for me the PCEngineCD and I want to add that logo so I don’t have to use a theme I don’t like.

A similar theme that is a harsh yellow that I do not like does have the correct emulator logo. I planned to just drag that image over.

Is there any other way to add a logo to a theme outside of FTP with ArkOS ?

Perhaps there is a **FEATURE REQUEST** here for access to Themes image files in the EASYROMS folder as most themes are missing one system or another.  I have yet to find a theme that is complete, especially when dealing with fringe systems.

christianhaitian, 2020-12-24:
What is your dongle?

OriginalNotmatthew, 2020-12-24:
Hello, thanks for the fast response.

It's the dongle that came bundled in with the unit.  No markings other than wifi type 802.11n

christianhaitian, 2020-12-24:
Are you able to scrape your roms or do online updates when using this dongle?

OriginalNotmatthew, 2020-12-24:
I am able to do both, although on the open scraper I did already max out my 30 day limit!

I update the OS everytiime you drop a change.  I update my cores everytime I update the OS.  I scrape missing nearly everytime I add a rom.

christianhaitian, 2020-12-24:
Nevermind, I reread your response.

Are you enabling remote services?

christianhaitian, 2020-12-24:
If you are, then it sounds like the wireless connection in your environment doesn't allow client to client communication.

OriginalNotmatthew, 2020-12-24:
I am, and I have toggled enable and disable multiple times.  I wait for the screen to run and show that it's enabled and I don't see an error (though the update text is really quick)

OriginalNotmatthew, 2020-12-24:
Interesting.  Is there a workaround?  Perhaps the default theme can be updated to have a logo for missing systems?  I had this same problem with Mame (cannot remember if it was 2003 or 2010 missing logo), "hacks" systems which are on some themes and not others, PC Engine CD, and Doom (it is in some themes and not others).  

I can of course get individual game images for each to work.

christianhaitian, 2020-12-24:
Ok.  I suspect it's a security setting in your wireless environment that doesn't allow client to client communication.  Are you able to ping your device from the computer your attempting to use to connect to the rg351p?

OriginalNotmatthew, 2020-12-24:
just "ping mydeviceIP" in terminal on mac?

christianhaitian, 2020-12-24:
Yep.  Try that.

OriginalNotmatthew, 2020-12-24:
56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
Request timeout for icmp_seq 5
Request timeout for icmp_seq 6
Request timeout for icmp_seq 7
Request timeout for icmp_seq 8
Request timeout for icmp_seq 9
Request timeout for icmp_seq 10
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12
Request timeout for icmp_seq 13
Request timeout for icmp_seq 14
Request timeout for icmp_seq 15
Request timeout for icmp_seq 16
Request timeout for icmp_seq 17
Request timeout for icmp_seq 18
Request timeout for icmp_seq 19
Request timeout for icmp_seq 20
Request timeout for icmp_seq 21
Request timeout for icmp_seq 22
Request timeout for icmp_seq 23
Request timeout for icmp_seq 24
Request timeout for icmp_seq 25
Request timeout for icmp_seq 26
Request timeout for icmp_seq 27
Request timeout for icmp_seq 28

still going.

christianhaitian, 2020-12-24:
Yep, assuming you got the IP from the network info option, your wireless network is blocking client to client communications.  If this is your personal home, is this a feature you know how to turn off or wish to turn off?

OriginalNotmatthew, 2020-12-24:
so in looking at the theme itself on github, I wonder if the issue I am trying to solve (I don't care about FTP other than this one thing at this moment) is just a folder naming convention. 

EASYROMS I believe has pcenginecd.  The theme is missing that system default, but the theme does support pce-cd as a folder name.

OriginalNotmatthew, 2020-12-24:
So I did work out a workaround that was easy.

I went to game collection by theme.

I put all the PCEngineCD games into PCE-CD on the theme

I sorted theme by system

Then I put HIDE on for PCEngineCD. 

NOW IT WORKS JUST LIKE I WANTED!

christianhaitian, 2020-12-24:
Ok.  Enjoy!

closed 2020-12-24 by christianhaitian