christianhaitian/arkos #116: Drastic won' t let me see what i'm doing!

crazyjoe29, 2020-12-25:
I've been trying to play tales of innocence which is a English patched ROM for nds which I found on cdromance. Com and I can't seem to be able to see what is happening in the game because all I can see is a blank screen not the game itself as it just doesn't seem to show any of the top screen and all it let's me see is the bottom screen and not the other one at all when I try to play it. Also the game music plays while I am trying to play the game. Is there a way to fix this, I can't seem to get it to play it right, what do I have to do to make it show the other screen so I can get farther in the game?

christianhaitian, 2020-12-25:
Per the FAQ:  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#a-

For Drastic (NDS emulator)

L2: Screen swap between upper and lower DS screens
R2: Swap between dual screen and single screen view
Right Joystick: Move the stylus
R3: Stylus tap
L3: Drastic menu
To exit the emulator, do so through the Drastic menu.

crazyjoe29, 2020-12-25:
Yes, I have known this already but it's not just a knowing how to use drastic  problem it's a not knowing how to get drastic to play/emulate my game/Roms right problem. I can't see the top screen for the game I mentioned in my earlier query even if I switch it to the top screen and I am not sure why but I have not been able to fix this, the game music will play but you can't ever see what you are doing because the top screen is blank and not showing any of the game play, plz help me fix this,thanks.

christianhaitian, 2020-12-25:
If it's not something the instructions above address, I can't help anymore because that particular emulator does not have sources available yet for reviewing issues and making improvements where possible.  What we have with that emulator is what we get until sources are available.

closed 2020-12-25 by christianhaitian

crazyjoe29, 2020-12-25:
thank you for telling me this, I did not know that. I was in a mind set of
that it could be enhanced to where you can play nds games better than it
was before. If that is the case, I won't ask you to see if you can at least
enhance the emulator's way of how it plays the game, I guess I have to
delete my favorite game since it can't be able to be played, sorry I asked.

christianhaitian, 2020-12-25:
No probem with asking.  How else would you know unless you ask and research.  Take care.