christianhaitian/arkos #117: Feature request: flappy birds, sudoku, tetris ports wanted

crazyjoe29, 2020-12-25:
I think that these 3 ports would be a great addition to the arkOS ports games as these are popular games or were as I am not sure if flappy birds would still have a popular demand for it but I think these games would be easier to emulate than the usually asked for ports and these are still very fun games in my opinion, what's your thoughts on these port suggestions? Please do let me know if you think that this could work or not and if you can do it. Thanks again for your help.

christianhaitian, 2020-12-25:
Are there particular sources for these?  As for tetris, there's so many versions available for emulators, why port a specific version if available?

crazyjoe29, 2020-12-26:
Yes, there are multiple upon multiple sources for these ports and all it takes is a few Google searches to find them and as for Tetris I think it a colored version is and would be very good to have because it is still a fun and the popular game is still high in demand and also these set of ports would really set your OS apart from other OS's (like emuelec for example). Also the flappy bird game community has been wanting a return of the game for a long time and to do this would make you their hero for bringing it back like this. Sudoku games are everywhere on the internet and wouldn't be very hard to find a suitable candidate to port, so if you can do it, please do, I think it would make many people happy, not just me. Please at least give it a few tries, OK.

christianhaitian, 2020-12-26:
Well differentiating and associating this OS is not a focus or concern for consideration of ports.  More so if the ports are of value to the community at large.  If someone recommends some specific source code for these, I can review what the possibilities for a port are and provide feedback when possible.

christianhaitian, 2021-01-10:
I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

closed 2021-01-10 by christianhaitian

crazyjoe29, 2021-01-10:
yes I do, but why don't you want to do ports anymore?

christianhaitian, 2021-01-11:
It takes to much time to figure out the resource needs for each port, increasing requests for ports  and it’s something I’m not really interested in.  My primary interest is emulation.  It’s why I got into this.  Now there are others like krishenricksen who seem to have a better handle on porting than I do so it’s best to look there.  The 351Elec community also has more ports and a community that has more developers there to assist with adding more.  You may want to look at their work in this area as well.

crazyjoe29, 2021-01-11:
i am sorry, i didn't know nor realize that and i am very grateful for your
work in emulation and I meant no disrespect to you and your amazing
project. also  I chose your project and not 351elecs, because of that very
same reason and I have no interest in switching because your's is so great
at emulation. Also I was only asking as to why you chose to no longer wish
to port games and now that I understand your reasons why, I won't ask
again. Plz keep up the good work and inspiring us all with all your
determination and hard work on this great os. And again I am sorry if I
insulted you.

christianhaitian, 2021-01-11:
You’ve done nothing wrong.  It’s just a decision I had to make after some realizations on my end.