christianhaitian/arkos #119: Problem with the latest version of PPSSPP

4n0nym0us, 2020-12-26:
I have been able to check the problem for both the latest ArkOs updates and the 351elec project, it has been updated to the latest version of PPSSPP. In both systems with our rg351p there is a bug that makes the games work in jumps, negatively impacting the fluidity of the video game. I am currently using ArkOs version 1.0.5 without updating, to avoid the problem. Obviously the problem is with PPSSPP. It is easily visible in titles like Daxter or Ridge Racer 2. Merry Christmas and thank you for ArkOs!

christianhaitian, 2020-12-26:
Just use the ppssppgo version available in the latest 1.5 update.

christianhaitian, 2020-12-26:
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-change-emulator-cores-in-ArkOS

closed 2020-12-26 by christianhaitian