christianhaitian/arkos #124: Wifi module inserted > No wake from sleep

markusveralius, 2020-12-27:
When I have an external wifi usb module plugged in and use the sleep function, the rg351p never wakes from sleep. However, if I turn on the machine without the wifi module, sleep seems pretty reliable.

christianhaitian, 2020-12-27:
Hmmm.  What wifi module are you using?

markusveralius, 2020-12-27:
Thank you for the reply Christian.

I no longer have the packaging, but it's a mini usb model from Asus,
produced 2019.

Cheers,
Mark

christianhaitian, 2020-12-27:
Sometimes with wifi in and active when the device is put to sleep, it takes a few seconds to wake from sleep.  Did you wait at least 10 seconds after pressing the power button for it to wake?  Also, did you have power plugged in at anytime while using the wifi usb module?

markusveralius, 2020-12-27:
Thanks for the suggestions. This is happening without the power plugged in
- it's just running on battery. I just did the 10 second test. It still
doesn't wake even if I wait much longer.

It happens when using Emulation Station as well as when in-game. The green
power light remains on, but the OS does not come back to life. I have
tested with both left and right-side OTG connections, and the same thing
happens.

I'm using the latest update of ArkOS (12/26), but this happened using
previous versions, too.

Here's a video to show you what I'm doing:
https://drive.google.com/file/d/17EwFoWLZNSHMBFFUbDuyChsSFfY64yQN/view?usp=sharing

Mark

christianhaitian, 2020-12-27:
strange.  Do you have another compatible wifi dongle you can try and see if it happens with it as well?

markusveralius, 2020-12-27:
Unfortunately not. Of course the workaround is simple - I only insert the
wifi dongle when I want to check for updates or scrape for rom info.

I did just buy Pico-8 and Voxatron for $19.99 and I'm going to check out a
few of the .png games with Pico-8 now.

Thank for the hard work and great OS!

Merry Christmas,
Mark

domramsey, 2020-12-27:
I'm having exactly the same problem. As above, it's not a huge issue when I can just remove the dongle, but the symptoms are exactly the same for me. I'm using this dongle:  https://www.amazon.co.uk/gp/product/B07FX549DS/

christianhaitian, 2020-12-27:
I did some further testing as I have a similar wifi module and confirmed this issue.  I have a potential fix that may address this.  It worked for me but before I push it out to others, I'd like for both of you to try it first and be sure it resolves this sleep issue for you as well.  See below:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test connectivity.  

You should hopefully see a good improvement in sleep stability and maybe wifi performance as well.  Let me know whether you do or not.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5744657/arkosbackup.tar.gz)

markusveralius, 2020-12-27:
I tried the fix, thank you. At first I had an issue with connecting to
wifi, and restarting several times to no avail. Then I deleted my wifi
connection and set it up once more in Emulation Station.

After that, it's working fine. The wifi dongle light is far more active
than before, constantly flashing.

I just checked the sleep mode with the wifi dongle. It seems to be working
great now!

christianhaitian, 2020-12-27:
Woohoo!

christianhaitian, 2020-12-27:
Needing to delete your existing wifi connection and set it up again is different but seeing as it was created with the less stable driver, it somewhat makes sense if the settings it used previously was different.  Thank you for confirming!

domramsey, 2020-12-27:
That fixed it for me too! Thank you!

christianhaitian, 2020-12-27:
And we're 2 for 2.  Thanks!  closing the ticket and will incorporate this fix for this chipset in a future update.

closed 2020-12-27 by christianhaitian