christianhaitian/arkos #129: Arkos stuck in boot loop after update (28/12/2020) and selecting ES-THeme-dosirakf

startreksteve, 2020-12-28:
OK, I installed a fresh install of Arkos 1.5, copied my roms across, all working fine did the latest update available today, all working fine again, chose the theme ES-Theme-dosirakf, and when emulationstation restarted to activate the theme, it got stuck in a boot loop that I couldnt escape. The theme worked fine before the update. Can I remove the theme?

christianhaitian, 2020-12-28:
Hmmm. I tried to replicate this issue and this doesn’t happen to me on my daily driver or a fresh install.  Hopefully you have a good name brand SD card?  You can delete the theme folder from the roms/themes folder.  It should be called es-themes-dosikraf.

christianhaitian, 2020-12-30:
Any update on this issue?

christianhaitian, 2021-01-03:
Closed due to no further updates on the issue.

closed 2021-01-03 by christianhaitian