christianhaitian/arkos #130: Feature request: NEO GEO CD support/documentation

fry-guy, 2020-12-28:
### Describe the Issue (If applicable)

Neo Geo CD is included as a folder in easyrom and the games will play, but there is no information about the required roms in the github documentation. Is it officially supported? If not, could it be, please? And if it, could you please add some official documentation?

### How can the issue be reproduced? (If applicable)

N/A

### What device are you using?

RG351P

### What version of ArkOS are you on?

I last updated 12/27/2020

christianhaitian, 2020-12-28:
Nice catch.  This has been added now.  See [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#neo-geo-cd)

closed 2020-12-28 by christianhaitian

fry-guy, 2020-12-28:
Awesome!