christianhaitian/arkos #131: Feature request

JohnConnor1978, 2020-12-28:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-12-28:
I may look into this in the future.

crazyjoe29, 2020-12-29:
I would like to have this feature also because it would be very beneficial to be able to move my ROMs or anything else needed for the handheld into the correct folders without having to keep taking out the micro SD card and wear down the contacts in the process of moving everything. I would like to see if you can also make a feature for the OTG port to allow us to play ROMs or whatever off a USB to micro SD reader from that port, to save space on the internal micro SD card for saving our games and to let us be able still play more roms without taking up too much said space would be amazing. With that feature you could hook up a hard drive of games and not having to worry about having to reflash and move all your ROMs again onto a bigger micro SD card which is very,very time consuming to do. I think that this is a good idea, what do you think?

fabinhojcr, 2020-12-31:
This would be a really cool and very useful feature, the developer is already doing an incredible job since he has almost no help from anyone.

christianhaitian, 2021-02-13:
This has been accommodated with today's update.  See FAQ [#23](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-mount-a-usb-drive-for-copying-and-moving-files) on how to use this feature.

closed 2021-02-13 by christianhaitian