christianhaitian/arkos #134: Install hangs on "System will reboot and load Emulationstation now after you hit the A button." I press the A button and nothing happens.

ajbowling, 2020-12-29:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2020-12-29:
It's ok to Just reboot manually when that happens.

ajbowling, 2020-12-29:
Okay, thanks! I did that and now it states that no gamepads detected. Advice? Thanks so much!

ajbowling, 2020-12-29:
Awesome thanks. Now the os boots, but the EasyRoms drive does not have folders in it?

christianhaitian, 2020-12-29:
No rom folders?  That shouldn't have happened.  Are you looking at this from your computer?

christianhaitian, 2020-12-30:
Any update on this issue?

ajbowling, 2021-01-02:
I reflashed the card and everything is great!

closed 2021-01-03 by christianhaitian