christianhaitian/arkos #135: Have a problem with file manager

Deer-Cr, 2020-12-29:
I have update the latest version 12/27/2020-1,but the file manager in 351p still have some issues. Dpad is useful,but every button(a,b,x,y,l1,l2,l3,r1,r2,,r3,select,start) except Dpad just repeats the last button of Dpad I've pressed. Something wrong with my351p?

christianhaitian, 2020-12-29:
Does this occur during game play as well?

Deer-Cr, 2020-12-29:
not, it occurs only in file manager.I shot a video here to describe this issue. [https://drive.google.com/file/d/1MRCiIrZBBI-LHZeKCMRDLzcJ-KGMoIT_/view?usp=sharing](url)

Deer-Cr, 2020-12-29:
not, it occurs only in file manager.I shot a video here to describe this issue. https://drive.google.com/file/d/1MRCiIrZBBI-LHZeKCMRDLzcJ-KGMoIT_/view?usp=sharing

christianhaitian, 2020-12-29:
That's because it's missing a configuration.  Something must have not completed with the update for it.  This should fix the issue:

Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

Once completed. Reboot you device and launch file manager again.  It should have proper controls now.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5750037/arkosbackup.tar.gz)

Deer-Cr, 2020-12-29:
Awesome! It can work properly now! Thanks sincerely

closed 2020-12-29 by Deer-Cr

christianhaitian, 2020-12-29:
You're welcome.  Enjoy!