christianhaitian/arkos #138: Feature Request: Port of Streets of Rage Remake

Qreg1984, 2020-12-29:
### Before submitting your issue or feature request

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

Qreg1984, 2020-12-29:
ArkOS is easily one of the best handheld OS' I've ever used.   Would love to see Streets of Rage Remake ported over. It has always been one of my favorite parts of these emulation handhelds and it has been properly ported to the Pi as well.

christianhaitian, 2021-01-10:
I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

christianhaitian, 2021-08-02:
I recently got streets of rage remake running for ArkOS.  If you'd like to help test it, can you tell me what your unit is?

Qreg1984, 2021-08-02:
I'd be happy to.  I'm using an RG351P.

christianhaitian, 2021-08-03:
Attached is a test version of SorR for ArkOS for the RG351V but feedback I've received says it works for the RG351P/M units as well.  Due to questionable legality, you'll need to provide your own sorr.dat file, mod folder, and palettes folder.  Below are the instructions you'll need to follow:

1. Take the linked compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder.  Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore ArkOS Settings, hit A.

3. Once completed, shutdown the system and remove your ArkOS SD card and put it in your computer.

4. Look for v5.1 of SoRR (**NOT VERSION 5.2 AS IT HAS SLOWDOWN ISSUES ON THIS SETUP**), and you can use either the windows or linux/debian version it does not matter which. Once you have the files, you will extract them to /roms/ports/sorr.  You should have a mod folder, a palettes folder, and SorR.dat file in this location in order for the game to work.  

5. Once completed, remove your ArkOS SD card and put it back in your unit and power it on.  Then go to Ports, and launch SorR.

**Important Notes** 
- DO NOT DELETE THE EXISTING SAVEGAME FOLDER OR THE SAVEGAME.SOR FILE IN IT! The save game file is what will set the correct controls needed.  
- Do not set the controls to anything but keyboard or the port will not function correctly.  Controls will spaz.  
- You can change the assigned keyboard values to a preferred key assignment of your choice using the dpad and action buttons.  
- You can exit the port at anytime by holding the select key for 1 second and pressing start or use the Select key when at the SorR main menu.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6920591/arkosbackup.tar.gz)

christianhaitian, 2021-08-05:
Have you had a chance to test this yet?

Qreg1984, 2021-08-07:
I did.   Got it installed perfectly using your guide and so far it plays
very well. I'm able to change the input setup and they all buttons work.
Audio sounds great.

There are some small spots of slowdown when there are many sprites on
screen at once but it's always temporary and doesn't effect gameplay much.

The only technical problem I've really run into is when the device goes
into standby (either from pressing the power button or just timing out) it
causes issues.  Sometimes the game will be frozen and sometimes it will be
running but the inputs have no effect when pressed.

All together it runs exactly like it should except for that one minor
thing.  Thanks again for putting the time into the port and letting me try
it out.  I'll put some more time into it and let you know if anything else
stands out.

tangent3D, 2021-11-09:
SoRR (tested 5.1) appears to be running great on the RG351MP, but controls aren't working yet.

christianhaitian, 2021-11-09:
They work just fine for me on my 351mp.  Try a reinstall from portmaster

tangent3D, 2021-11-09:
Thanks very much! I followed an out of date tutorial. That worked perfectly and I'll be sure to try PortMaster first from now on.