christianhaitian/arkos #139: Feature Request: Hide "ArkOS Bar" when screensaver is running

Disfunktional, 2020-12-29:
Is it possible to have the ArkOS bar at the top of the screen hide when the screensaver is running?

Transferring files or scraping can sometimes takes a while, and just feel that over time that creates risk of screen burn. If the bar was hidden when the screensaver is running that would mitigate the risk.

christianhaitian, 2020-12-29:
This is not something that I will work on.  Good news is that the rg351p, rk2020, and rgb10 have IPS screens and they don't get burn-in. However, you could get image retention which is a temporary effect.  If you're concerned about image retention or still a possibility of burn in, you can bring the brightness down to 0 using the global hotkey for controlling brightness (r3 + dpad down).  This has the benefit of reducing heat generated from the panel and saving the life of the panel when you're doing these activities.

closed 2020-12-29 by christianhaitian

Disfunktional, 2020-12-30:
All good, thanks for responding Christian