christianhaitian/arkos #140: Support for 3DO (Opera) emulator

ZoranLain, 2020-12-30:
I really like ArkOS on my RG351P.  The thing I miss the most is support for 3DO emulator Opera.  I would really appreciate if you can add this in a future release.  Thanks.

christianhaitian, 2020-12-30:
I haven't tested and confirmed myself but from what I've heard, 3DO will load but not run well.
If you can assist, you can try the opera 64 bit libretro core through the retroarch64 menu and see what you get.  Just go to retroarch->Online Updater->Core Downloader->then select opera.  You'll need to have the appropriate 3do bios files in the bios folder and a game or two to test.  More info about it [here](https://docs.libretro.com/library/opera).  Let me know how your testing goes.

ZoranLain, 2020-12-30:
Thanks for looking into this.  I just did some tests based on a few games on Retroarch64.  I did an ArkOS update & core updates within retroarch before I started.
1) Captain Quazar - Intro perfect video & sound (@54~fps).  Gameplay slow but not crawling (@21 to 24fps).  Sound was stuttering in game.  I would say not a good experience.
2) Gex - Intro not bad with some crackling sound (@44~fps).  Gameplay is 26  to 28fps.  I would say playable but not enjoyable.
3) Night Trap - Intro almost perfect (@46 to 55 fps), minor sound crackling almost as if the vol is turn up too high if you know what I mean.  Gameplay performance is great like the intro.   I would say this one runs great.
4) Crypt Killer - Intro perfect video & sound (@58~ consistently).  Gameplay is great as well (@44fps under heavy action). I would say this one runs great.
After the testing, I think the FMV games are going to run perfect.  The other game types are questionable.   This doesn't bother me so much because the games I want to play are the FMV games.   I know they are cheesy but I love them =p

christianhaitian, 2020-12-30:
Ok.  Thanks for testing.  I'll add support for this emulator in the next upcoming update.

ZoranLain, 2020-12-31:
Great!  Thanks.   I really appreciate this.  Happy new year!

christianhaitian, 2021-01-03:
This has been accommodated with today's update.  Just update from options section

closed 2021-01-03 by christianhaitian