christianhaitian/arkos #141: Diablo freezes when in town around level 20+ characters.

rayhern, 2020-12-30:
You may have to talk to some people or move things around in the inventory to trigger the freeze. I am not sure if it is a memory leak or what. Let me know if you need anything from me.

[single_0.sv.zip](https://github.com/christianhaitian/arkos/files/5755422/single_0.sv.zip)

rayhern, 2020-12-30:
Is there a way in Arkos to create a huge swap file before boot? That way I can be sure its not running out of memory? Thanks in advance.

christianhaitian, 2020-12-30:
You can try these [instructions](https://docs.fedoraproject.org/en-US/Fedora/14/html/Storage_Administration_Guide/s2-swap-creating-file.html) for creating a swapfile and see if it solves this issue.  Let me know your results.

rayhern, 2020-12-30:
I changed my Devilutionx.sh to the following:

```
#!/bin/bash
dd if=/dev/zero of=/roms/swapfile bs=1024 count=1024000
mkswap /roms/swapfile
swapon /roms/swapfile
export SDL_GAMECONTROLLERCONFIG="$(cat /roms/ports/devilution/gamecontrollerdb.txt)"
/home/ark/.config/devilution/devilutionx
unset SDL_GAMECONTROLLERCONFIG
```

It created the swap file 1gb in size, and it took a while to start. Diablo then froze as soon as the game was loaded so this approach did not work. Here is my discussion with the devilutionX team...
https://github.com/diasurgical/devilutionX/issues/915

christianhaitian, 2020-12-30:
So it's an issue upstream then.  Ok, will monitor progress on this.

rayhern, 2020-12-31:
I just wanted to keep you in the loop, it may still be too early to tell.. But I think I have a solution. Here is my new Devilutionx.sh:

```
#!/bin/bash
killall emulationstation
export SDL_GAMECONTROLLERCONFIG="$(cat /roms/ports/devilution/gamecontrollerdb.txt)"
/home/ark/.config/devilution/devilutionx
unset SDL_GAMECONTROLLERCONFIG
emulationstation
```
So basically I force close emulationstation, before launching devilutionX, and launch it again after. Not sure if it's working because it has more resources, or if something was conflicting with SDL and emulationstation...Either way fingers crossed this works for a while, only time will tell. Have a good one.

christianhaitian, 2020-12-31:
Interesting. Depending on further testing, I wonder if this may help with reports of half-life crashing randomly as well.

rayhern, 2021-01-01:
Diablo/Devilutionx is still freezing. Some good news is the devs think they found the problem. It should be patched in 1.2.0

christianhaitian, 2021-01-01:
Cool.  Thanks for the update on this.

rayhern, 2021-01-06:
I found the solution finally to Devilutionx, Long story short make sure you use the GOG diabdat.mpq, and newest patch_rt.mpq. That is the answer. The version I had was freezing talking to Pepin, or just in general when my character level reached over 20. Now I have started the game multiple times, and its not freezing at all. Thank you for all your help with this issue. I don't know why my original mpq was messed up, but I am sure its happening to others.

christianhaitian, 2021-01-06:
Very nice.  I’ll add this to the Emulators and Ports section so others are aware.  Thank you for the follow up on this.

closed 2021-01-06 by christianhaitian