christianhaitian/arkos #143: Update error

fabiocattaneo, 2020-12-31:
I’ve flashed the ArkOS image linked in the main wiki page, performed the first setup correctly, connected the WiFi dongle and checked the network status. Then performed a system update. It went fine until an error occurred: cannot resolve “GitHub”. Tried again, no luck

christianhaitian, 2020-12-31:
The url is correct.  The issue is most likely the wifi adapter has stopped responding. Reboot your device and try again.

fabiocattaneo, 2020-12-31:
I already did it 4 times, but it always stops there. Once returned to the ArkOS home, the WiFi is still working...

christianhaitian, 2020-12-31:
According to the picture you posted, you're unit is not able to resolve github.com at all of a sudden.  Are you able to do anything else online with the unit like scrape game thumbnails or videos?

fabiocattaneo, 2020-12-31:
Yes, I’ve tried exactly that and it worked... I think it could be some sort of issue with the chipset of my WiFi dongle. Power saving is off, by the way.

christianhaitian, 2020-12-31:
My guess is that your wifi dongle is based on the rtl8188 or rtl8192 chipset.  Take a look at a possible fix for this [here](https://github.com/christianhaitian/arkos/issues/124#issuecomment-751475195).  Can you try this then attempt to update again and let me know what you get?

fabiocattaneo, 2021-01-01:
You're right, it's based on the rtl8192cu chipset.
I've installed the fix and I'm happy to confirm it worked!
Thanks a lot :)
The console still makes weird noises when the wifi dongle is working, but now the network is faster and much more stable.
Would it be possible to include the fix in future updates?
Anyway, happy new year!

christianhaitian, 2021-01-01:
It's already included in the update after the one you were getting stuck at. LOL.  The next image will include that built in.  Happy New Year to you as well!

closed 2021-01-01 by christianhaitian