christianhaitian/arkos #151: Colecovision emulator

Mika73, 2021-01-02:
I can’t get Colecovision games working. It just starts MSX Basic. It seems that I am not only one with this problem other people had this problem at Facebook groups and Internet forums. 

I have updated my ArkOS and using RG351P. 

Have copy BIOS folder from original sd card and download Coleco bios file from internet.

christianhaitian, 2021-01-02:
Has this been completed per the note section of Emulators and Ports section for Colecovision?

Notes: The blueMSX core requires the 'Databases' and 'Machines' folders from a full installation of blueMSX.
You can download the 'Databases' and 'Machines' folders from [an official full standalone blueMSX](http://bluemsx.msxblue.com/download.html) emulator installation.
Get blueMSXv282full.zip near the bottom of the page.
Move/Copy the 'Databases' and 'Machines' Folders to the bios folder.

Mika73, 2021-01-02:
Hi

Those were already at Bios folder at original sd card. I did download stand
alone version of BlueMSX and those folders/files were same. No help. It
just start MSX Basic.

Best regards
Mika

christianhaitian, 2021-01-03:
I was able to test and confirm that Coleco vision loads up just fine on my end when I copied the Databases and Machines folders from blueMSXv282full.zip to the /roms/bios location and a good coleco.rom bios to the bios folder as well.  I was able to successfully load Bump 'N' Jump and Boulder Dash.

closed 2021-01-03 by christianhaitian

Mika73, 2021-01-03:
Thank you. I actually got it working. Problem was with some Coleco game roms that work with real Colecovision, but emulator boot those msx basic.