christianhaitian/arkos #162: Feature request : Daphne from Batocera

TigTog, 2021-01-04:
Loving ArkOS !  For me, the addition of Daphne emulator from Batocera would be awesome. It runs great on the RG351.  

Hope you can find time to make my day! 

Dale

christianhaitian, 2021-01-04:
Already in the works.  No ETA yet

christianhaitian, 2021-01-10:
This has been accommodated with today's online update.  

* for key assignments, see Daphne section within the rg351p global hotkey section of faq 
    (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RK2020#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS)

* for rom loading, see emulators and ports section (https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#daphne)

closed 2021-01-10 by christianhaitian

TigTog, 2021-01-10:
Thank you SO MUCH Christian.  You have made my RG351P the best device i've every owned.  The hardware is great and you have made the software 1000% better.   I simply cant thank you enough.  I'll be walking around today with a huge smile on my face !  Stay Safe. Dale.