christianhaitian/arkos #163: Building the new devilutionx worked, but i am missing some functionality in the controller.

rayhern, 2021-01-05:
I build everything needed for devilutionx on the device itself using ssh, then I copied it over to the correct .config location of the original binary. Is there anything else special I have to do with the build flags? Thanks for your help.

christianhaitian, 2021-01-05:
If you built the 64 bit version, make sure the mouse still works.  Otherwise, you may have to build it in 32 bit like I had to to get mouse controls working.

This is the cmake options I used for devilutionx.  

-DNONET=ON -DCMAKE_BUILD_TYPE="Release" -DASAN=OFF -DUBSAN=OFF -DDEBUG=OFF -DLTO=ON -DDIST=OFF -DFASTER=OFF -DPREFILL_PLAYER_NAME=ON

rayhern, 2021-01-05:
export CFLAGS=-m32    is this how you built for 32 bit? Thank you so much for helping me out as always I will share my findings with you.

christianhaitian, 2021-01-05:
You can try that but I do all my builds in chroots using a virtual dev environment I built.  If you 're interested in trying it out, see here: https://forum.odroid.com/viewtopic.php?p=306185#p306185

rayhern, 2021-01-05:
Wow thanks, I got the virtualbox installed and when I double click arm32 on the desktop it doesn't have internet access. When I use the regular terminal it does. Does this happen to you? I have to download the requirements.

rayhern, 2021-01-05:
The more I look into the bug reports it seems the problem may be related to sdl2. I am still looking into it. I tried updating sdl2 on rg350 and I got a never-ending rumble when I booted the device. So that is out of the question.

christianhaitian, 2021-01-05:
Internet works just fine for me in the chroot however, someone else told me they had to connect Outside of the chroot then move the resolv.conf from /etc to the chroot /etc folders To get internet to work on their end.

rayhern, 2021-01-05:
Thank you so much for all your help. I was able to build the latest devilutionx from the 32 bit chroot.

christianhaitian, 2021-01-05:
Nice.  Let me know if it fixes the crash issue.

rayhern, 2021-01-06:
I have tried all day with different compiler flags. The issue is still happening on level 20+ characters. (freezes in town) I will let you know once I have a winning combination. Thanks again for all your help, I really appreciate it.

closed 2021-01-06 by christianhaitian