christianhaitian/arkos #166: Feature request : Delete rom file at Game Options

fabinhojcr, 2021-01-05:
It would be great to have the option to delete the games I no longer want in the game options.

SRGilbert, 2021-01-05:
You can already do that.  Highlight the game, hit Select to open the Options menu, and choose Edit This Games Metadata.  Delete button is at the bottom.

fabinhojcr, 2021-01-05:
But that only excludes it from the list of games, I would like it to be permanently deleted from the memory card.

Xiscf, 2021-01-05:
Hi

In that case, open the file manager and delete the game from there.

fabinhojcr, 2021-01-05:
But generally arcade games usually have confusing names and often I don't know what games they are, in the list of games I can know what the game is and delete it correctly if there is this option.

SRGilbert, 2021-01-05:
> But that only excludes it from the list of games, I would like it to be permanently deleted from the memory card.

No, it definitely deletes the rom.  I just tested it by deleting a game and then going into the file manager to confirm.

Xiscf, 2021-01-05:
I see.
I thought it was possible to get the name of the rom inside the metadata.
What about displaying the name of the file instead of the name of the game?  Then with the file manager you would be able to find the good one.  Could it work?

Xiscf, 2021-01-05:
GG @SRGilbert  :)

closed 2021-01-05 by christianhaitian