christianhaitian/arkos #170: I’m  having an issue

Bobbyculp, 2021-01-06:
So after lots of research  I have to just be humble and ask so i this is is feat and is the most user friendly but here comes my question I how do I save games i was able to figure out everything else out I have been playing Ark the lad and the in game save says it’s saving but when I restart I and select continue it  just starts Over I know I am probably missing something very simple thanks

christianhaitian, 2021-01-06:
When you're done playing the game, how do you exit?  Assuming you're using the rg351p, you have to make sure you exit by hitting start+select.  Then the game save is written to the sd card.  If you're just shutting down the device when done playing, your saves will not be there.  Also, as an additional backup plan, you should use savestates as well.  This way if something is wrong with your save, you can at least restore your savestate.

Bobbyculp, 2021-01-06:
Thanks I will do

closed 2021-01-06 by Bobbyculp