christianhaitian/arkos #172: Bug report:3do/jaguar lost

duskgao, 2021-01-06:
I found 3do/jaguar menu in EmulationStation had been losted after 0103‘ update.

There is no 3do/jaguar content in es_systems.cfg also.

Should we fix it soon?

christianhaitian, 2021-01-06:
3do and jaguar are still showing available on my dnd across 3 different sd cards that I use to test the updates.  Do you modify your es_systems.cfg file at all?

duskgao, 2021-01-06:
Nope, I know what es_systems means,so I definitely leave it alone ;) ... there are 4 es_systems in the dir. One is es_systems.cfg,there other are es_systems.cfg.update010X2021...all of them haven't 3do/jaguar parts...

Maybe It only found on my rk2020...never mind... close it. 
Great work! dude~~

closed 2021-01-06 by christianhaitian