christianhaitian/arkos #174: Feature Request: "Blood" port

wyterabitt, 2021-01-07:
I don't know if this will be possible, but thought I could at least ask about it. Would a port of Blood be possible for the 351p, using Nblood which apparently can be ported Linux. We would just need our own copy of games files. 

If it is not possible, too much work, or beyond the capabilities of the device I understand, and thanks for the work you do. 

(there is also Exhumed and Duke3d ports from the same project, but I thought they might be even less likely) 

https://www.blood-wiki.org/index.php/NBlood

http://nukeykt.retrohost.net/

https://github.com/nukeykt/NBlood

krishenriksen, 2021-01-10:
Duke3d can be added by installing https://github.com/krishenriksen/RockRiotMenu I will look into Blood for you.

christianhaitian, 2021-01-10:
Per posted above, you may look at this for a possible port.  I will no longer be focusing on adding additional ports at this time.  You may look for alternative unofficial solutions such as posted by kris above.  Do understand support for those ports will not be available on this github.

closed 2021-01-10 by christianhaitian

wyterabitt, 2021-01-11:
@krishenriksen  This looks like an awesome addition, and thanks for looking into Blood as well.

@christianhaitian  Thanks for letting me know, and thanks again for all the work you do man.

krishenriksen, 2021-01-11:
@wyterabitt

 Added the following games to RockRiotMenu

     Blood
     Exhumed
     NAM
     Redneck Rampage
     Redneck Rampage Rides Again
     World War II GI

 Simply enter menu, click `Update` and enjoy!
 
 @christianhaitian you can close this.