christianhaitian/arkos #176: Hi,

Jooppie, 2021-01-07:
Hi,
This happens to me also. I replaced the ports folder with the original folder and it is OK. However after scraping for metadata the menu is screwed up (3D-Demon on top now which starts Doom2).
So it reproducable (latest ARK 0S 1.5  05012021)
Ron

_Originally posted by @Jooppie in https://github.com/christianhaitian/arkos/issues/48#issuecomment-756203875_

Blobb56, 2021-01-07:
Has ScummVM also had its name changed?

christianhaitian, 2021-01-07:
Don't scrape the ports folder.  Those entries are custom and I doubt that they can be scraped correctly.  ScummVM has not had it's named changed at all.

closed 2021-01-07 by christianhaitian

Jooppie, 2021-01-08:
How can you prevent scraping this folder when scraping all?

christianhaitian, 2021-01-08:
when you hit scrape now, go to systems, hit the up button so it putts you at the bottom of the list and unselect ports.  then go back and hit select start to scrape.