christianhaitian/arkos #177: N64 Games Not Appearing

Parochus-glitch, 2021-01-07:
Running the latest update on RG351P as of this writing, I realized my N64 games weren't appearing at all in the menu. But when I booted them up directly through Retroarch they will then show up when the system is rebooted. I checked and the folder is named correctly. I had done a copy-paste from the folder structure of a 351Elec install I also have done, so I'm wondering if I copied something over that is causing this.

christianhaitian, 2021-01-07:
If those n64 games from 351elec had been scraped, try deleting the gamelist.xml file in the folder.  Otherwise, make sure the extensions are ones that are supported per this [link](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#nintendo-64).

Parochus-glitch, 2021-01-07:
Thanks. I had caught most of the gamelist.xml files in other folders but not N64. Unfortunately, deleting the file doesn’t update the menu. Anyway to force? Rebooting doesn’t change anything. (And I confirmed extensions are supported).

christianhaitian, 2021-01-07:
make sure the folder is named n64 and not N64.

Parochus-glitch, 2021-01-08:
Yup, that checks out. Weird.

Parochus-glitch, 2021-01-08:
Sorry, that might not have been clear. It was already “n64” and not “N64” so that’s not the issue unfortunately.

christianhaitian, 2021-01-08:
make sure the n64 system is a visible system in es.

Parochus-glitch, 2021-01-08:
It is. Just to be clear, the problem was that the system would not initially appear, but once I play a game in retroarch that game will now appear in emustation on reboot, but only that game and any others played that way. So it’s not automatically scraping the folder.

christianhaitian, 2021-01-08:
Is the system set to only load games parsed in a gameslist?  Check start>advanced settings>parse gamelist only.  This should be set to off.

Parochus-glitch, 2021-01-09:
Yes, that’s set to off.

christianhaitian, 2021-01-09:
Try removing your n64 folder.  Boot the unit up with the sd card loaded.  Shut it down, then recopy your n64 games to the sd card in the appropriate folder and make sure no gamelist.xml file is included.  See if that resolves the issue for you.

Parochus-glitch, 2021-01-10:
Okay, the problem was all the files were in .zip format. Retroarch was automatically decompressing the files when I played them, which is why they would show up. I assumed since zip files were working for other emulators that they should have worked in this case, that was a bad assumption, should have looked harder at the supported format page. Feel like a dummy. Thanks for your patience!

closed 2021-01-10 by christianhaitian