christianhaitian/arkos #182: Amiberry control with d-pad

MarcoVer77, 2021-01-09:
On the Odroid go advance 1.0 you can’t use Amiberry with the d-pad. Left/right is up/down.
Have tried to solved this within the menu but that’s not possible.

christianhaitian, 2021-01-09:
Yea those controls were a little of a mess.  Here's a fix that will work for the OGA 1.0 and RK2020 only.  This fix will also allow you to exit amiberry using select+start (f1+f6).  I plan to include it in a future online update as well.

See below:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device then test and confirm the button mapping has been fixed for amiberry.  

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5791545/arkosbackup.tar.gz)

MarcoVer77, 2021-01-09:
Nice, thank you very much!

closed 2021-01-09 by christianhaitian