christianhaitian/arkos #185: Feature request adding rumble to pokemon mini emulator

pentaenix, 2021-01-11:
Many games use this feature, it would be nice to be able to activate rumble (I've tried from the retroarch menu to no avail)

christianhaitian, 2021-01-11:
Cool idea to improve this emulation experience.  Here's a test version of the pokemon mini emulator core with vibration added for the rg351p. I tested it with Pokemon Party Mini and it seemed to work well but I am not very familiar with pokemon mini just yet but I'm reviewing it as it seems neat. I plan to include it in a future online update as well if it works well for you and others who may test this.

See below:

Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

Once completed. Reboot the device then test and confirm the vibration works well in pokemon mini and doesn't negatively impact the rest of the emulation and let me know.

Thanks
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5794251/arkosbackup.tar.gz)

pentaenix, 2021-01-11:
I've been testing it with many games, and it works great! Haven't found any issues, but if I find any ill let you know, thanks!

christianhaitian, 2021-01-11:
Excellent.  Much appreciated for your testing and feedback.

closed 2021-01-12 by christianhaitian