christianhaitian/arkos #186: I got multiplayer working with devilutionx and arkos!

rayhern, 2021-01-11:
Hello, 

Just wanted to let you know I got devilutionx working for multiplayer in ArkOS!

You can clone my repo here: https://github.com/rayhern/devilutionX-rg
(I had to make some multiplayer fixes that I just commit to my own repo.)

The same applies here for connecting to hosts as the RG350.

The user must edit diablo.ini as follows:

```
[Phone Book]
Entry1=192.168.0.2
Password1=rayhern
```

Just so you know I had to SSH to my device with ArkOS on it and run "sudo apt install libsodium-dev"
(So that it properly installed and linked the sodium library. I did the same in the buildroot (arm32))

After editing your diablo.ini the game will pre-populate the host and the password when both creating games
[devilutionx-arkos-multiplayer.zip](https://github.com/christianhaitian/arkos/files/5793957/devilutionx-arkos-multiplayer.zip)
 and joining them.

Just doing my part to help. Arkos is great! Keep up the good work. I love Ubuntu/Debian.

I have included the binary if you didn't want to compile it yourself. I am sure though you would prefer the explanation. Cheers.

BTW here are the build flags I used:

-DCMAKE_BUILD_TYPE="Release" -DASAN=OFF -DUBSAN=OFF -DDEBUG=OFF -DDIST=OFF -DPREFILL_PLAYER_NAME=ON -DBINARY_RELEASE=ON

rayhern, 2021-01-11:
[devilutionx-arkos-multiplayer.zip](https://github.com/christianhaitian/arkos/files/5793958/devilutionx-arkos-multiplayer.zip)

christianhaitian, 2021-01-30:
Thanks for sharing.  I don't have plans to support multiplayer devilution at this time but if I do, I might check it out.  Take care!

closed 2021-01-30 by christianhaitian