christianhaitian/arkos #191: Headerless NES roms

alpaga77, 2021-01-12:
where i can put NstDatabase.xml from Nestopia to properly play headerless roms?

christianhaitian, 2021-01-13:
Interesting.  Didn’t know anything about this until now.  It looks like if you put this in the bios folder, it should work.  Can you test this and confirm whether or not this works and report back?

alpaga77, 2021-01-13:
not working

christianhaitian, 2021-01-13:
How are you testing this?  What specific game or hack are you using?

alpaga77, 2021-01-14:
I have a problem because my friend broke my RG351P launch button. But I will try to describe the situations. As an example, I want to run a SPY HUNTER (EU) rom compatible with DAT-o-MATIC with CRC32 C528ED56 must have this CRC32 so that in the desktop version of Retroarch it can be automatically loaded into the playlist. On desktop version of Retroarch with Nestopia core it works fine. On RG351P it starts and before anything is displayed returns to roms list. I downloaded the rom from the internet, but in order to use it in the desktop version of Retroarch and load it automatically into the playlist, I had to download the program that removed header from the rom and then CRC32 changed into C528ED56 and then the desktop version of retroarch loaded the rom.

alpaga77, 2021-01-14:
desktop retroarch automatically loads rom to playlist if its CRC32 matches with DAT-o-MATIC but for CRC32 to match in most cases you need to remove header from rom. The NstDatabase.xml file (https://github.com/0ldsk00l/nestopia/blob/master/NstDatabase.xml) contains a list of what the mapper for the specific CRC32 emulator must use to run the game without header

alpaga77, 2021-01-14:
NstDatabase.xml for example

<game>
        <cartridge system="NES-PAL-B" dump="ok" crc="**C528ED56**" sha1="70E6CB9E37D27DFB65E49B69D542A60617D19632">
            <board type="NES-TLROM" mapper="4">
                <prg size="128k" />
                <chr size="128k" />
                <chip type="MMC3B" />
            </board>
        </cartridge>
    </game>

christianhaitian, 2021-01-16:
Try putting the file in /home/ark/.config/retroarch/system folder.  You can do this via sftp using winscp or filezilla when connected to the network following the instructions [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-ssh-into-ArkOS)

christianhaitian, 2021-01-19:
Any update on this?

alpaga77, 2021-01-20:
I can't check now because I have to solder a new power button. As soon as a new one comes, I will check it

christianhaitian, 2021-01-29:
Closing due to no further updates on the issue.  If you have any further updates on this, feel free to respond to this ticket.

closed 2021-01-29 by christianhaitian