christianhaitian/arkos #196: ResidualVm

mmoyano88, 2021-01-14:
Is there a way to play games on ResidualVm?  Could that function be added?

christianhaitian, 2021-01-16:
I started to review and the then read that Residual VM has merged with ScummVM as of October 10, 2020.  https://www.residualvm.org/news/20201010/
Standalone scummvm is already included in ArkOS.  Have you tried to run the particular residual VM game you are hoping to play using the standalone scummvm emulator?  As the standalone scummvm is not the default emulator, you'll need to set it as the main scummvm emulator for the system or per game using the instructions [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-change-emulator-cores-in-ArkOS)

closed 2021-01-16 by christianhaitian

mmoyano88, 2021-01-16:
Thanks for your time and attention.  I have tried what you say, to load the core and then the game, to search for the game (it does not appear in the ScummVm list) and it cannot load it.  I assumed that I needed to configure something in the core, or in the system, so I asked if that function could be added.  But it is not so urgent either.  Thank you

christianhaitian, 2021-01-16:
What is the game you attempted to load so I can test?

mmoyano88, 2021-01-16:
I have tried playing Grim Fandango.  All ScummVm games appear in the games list, but the ResidualVm ones don't, and from the file explorer I can't open it with the Scumm core.  When I open Scumm I can add games to the application, but with the Residual ones it gives an error.

One more time, thanks

christianhaitian, 2021-01-17:
So I got Grim Fandango running with the standalone scummvm emulator but it was tricky.  

1. I first copied the game folder containing all of the grimdata files to the scummvm roms folder on the micro sd card.  
2. Then I copied the gfupd101.exe file from [here](https://downloads.scummvm.org/frs/extras/patches/gfupd101.exe) and the grim-patch.lab file from [here]( https://github.com/giucam/residual/raw/master/dists/engine-data/residualvm-grim-patch.lab) into the same game folder containing the grimdata files.  
3. Make sure to rename the residualvm-grim-patch.lab file to grim-patch.lab or the game will not load.  It will just drop you into a debug mode.  
4. Once you've done all of this, you'll need to load the standalone scummvm emulator, add the game, go to the graphics tab and set the Game 3D renderer to software. 
5. Then start the game.

mmoyano88, 2021-01-17:
It works. Thank you very much