christianhaitian/arkos #198: Feature Request: Full ASCII WiFi Password Implementation

KarwoodBear, 2021-01-16:
### Describe the Issue (If applicable)
My WiFi password has a ^ in it. - also it appears & isn't a character you can enter too.. might be more!

### How can the issue be reproduced? (If applicable)
try to set the WiFi password.

### What device are you using?
RG351P

### What version of ArkOS are you on?
1.5.7

### Anything else you'd like to include that may help to review this issue or feature request?
Nope! Program is awesome!!

(and yes I know I can change my password, but I have a lot of devices!)

KarwoodBear, 2021-01-16:
As a workaround, I did find the psk hiding inside /etc/NetworkManger/system-connections in plaintext.

christianhaitian, 2021-01-16:
Already a known issue per item number 8 here https://github.com/christianhaitian/arkos/wiki/Known-issues.  Hope to address in the future but I have no timetable set on when this will be done at this time.

closed 2021-01-16 by christianhaitian