christianhaitian/arkos #199: connection issue with Unifi access point

KarwoodBear, 2021-01-17:
4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
(I promise I reviewed it this time 👍 )

### Describe the Issue (If applicable)
Using ArkOS, and a TP-Link USB WiFi Adapter TL-WN725N, ArkOS can't connect to the house internet system by Unifi, but can connect to many other networks (phone hotspot, tp-link access point repeater on the network that otherwise doesn't connect)

Using 351Elec or the stock firmware, there is no issue with connection and data transfer.

### How can the issue be reproduced? (If applicable)
The software sees the access point and attempts to connect, but does not seem to be able to make the connection. I haven't tried changing many of the parameters in the config file to try to work around it, though. I do understand this issue is probably dependent on my specific hardware.

### What device are you using?
RG351P with TP-Link USB WiFi Adapter TL-WN725N soldered to the motherboard (not using OTG)

### What version of ArkOS are you on?
v1.5.01.16.2021 (which just successfully updated over the access point repeater)

### Anything else you'd like to include that may help to review this issue or feature request?
Have a Ko-Fi or something?

christianhaitian, 2021-01-17:
There's a test image for ArkOS 1.5 with additional wifi drivers including Atheros ones.  I've received reports that it has stabilized being able to use wifi for the few that have tested it so far.  If anyone is struggling with reliable wifi on ArkOS, you're welcome to try this and provide any feedback you may have about it.

https://mega.nz/file/PcoBlAST#cN9oVCplJRu0nzqwtowP1-aJkJpEBnYRhKFBhni28M4

It was created on the 12/19/2020 so you will have to update it once you've installed it for the latest changes and if you follow the instructions here with it, it should stabilize it per the reports I've received.  https://github.com/christianhaitian/arkos/issues/124#issuecomment-751475195

christianhaitian, 2021-01-19:
Any update on this?

KarwoodBear, 2021-01-19:
RL came up this weekend, let me hammer at the emulator a bit this week?

christianhaitian, 2021-01-29:
Closing due to no further updates on the issue. If you have any further updates on this, feel free to respond to this ticket.

closed 2021-01-29 by christianhaitian

metaColin, 2021-06-24:
*Bump*

I am unable to connect to my Unifi wifi with my RG351v using the latest release (v1.7 - 06/04/2021).

In fact, ArkOS won't even save the Wifi credentials I enter. After selecting my network and entering credentials I am simply returned to the previous screen.

Sounds like it may be the same issue.

Some more points of data:

- If I switch to 351Elec the RG351v has no trouble connecting to the same network.

- My RG351p has no trouble connecting to the same network.

christianhaitian, 2021-06-25:
Try setting up the wifi using retroarch.  Should be in the settings>wifi section.

metaColin, 2021-06-25:
I can't find any wifi settings in retroarch, not in settings, not in network.

I even tried changing the theme to XMB thinking maybe that some themes didnt expose the wifi settings.

Is there a toggle to expose wifi settings in retroarch that I am missing?

christianhaitian, 2021-06-25:
If you’re truly on ArkOS 1.7 (6/04/2021), it should be available in retroarch (not retroarch32) as I described above.

metaColin, 2021-06-25:
I double-checked just to be sure, and it looks like I am actually using ArkOS 1.7 (01/15/2021).

I know the releases page says that there is (06/04/2021) but it looks like the actual download links for the RG351v on the wiki page the latest is (01/15/2021).

Compiling from the source on the release page is a little beyond my knowledge. Is there a version of 06/04/2021 precompiled for RG351v that I am missing? Is there someway to upgrade from the 01/15/2021 version without a wifi connection?

christianhaitian, 2021-06-25:
There is no new image.  Updates are only available via wifi at the moment.  Perhaps you can try to update via mobile hotspot suchas from a smartphone?

metaColin, 2021-06-28:
I managed to update with an ethernet adapter. Upon update, I had wifi settings in retroarch and was able to connect as you described.

Thanks.