christianhaitian/arkos #200: Feature request terminal with OSK

coremaven, 2021-01-17:
Would it be possible to add a terminal with an on-screen keyboard? It could be very helpful in having quick access to a terminal for tools like Nano/VIM. Plus having a portable Linux box in my pocket that has a terminal could be very helpful in countless ways.

Thank you.

christianhaitian, 2021-01-17:
It may be possible but not something I will focus on any time soon.  The goal of this distro is for emulation.  Best I can recommend is a small portable usb c keyboard to connect to the device while in terminal mode.  Not as portable but it would work.

closed 2021-01-17 by christianhaitian

coremaven, 2021-01-18:
Thanks for considering the idea and everything you have done for this distro. By far the best option for this device.