christianhaitian/arkos #201: No rumble after MGBA Update

lokisilverblade, 2021-01-17:
Hi,

i installed the latest update yesterday in hope Pokémon Pinball on GBA will work flawless with it. But in Emustation i choosed Auto Core ore Rumble MGBA but that have no effect. The option in game is outgreyed. My Device is RG351P

Thank you, and thank your for your great work.

christianhaitian, 2021-01-17:
I was able to play this game with rumble working without issue on my end.  Did you do any updates via retroarch after you did the ArkOS update?

lokisilverblade, 2021-01-17:
Hallo, no only the updates with the arkos Update script.

lokisilverblade, 2021-01-17:
Okay after fresh install no rumble option in mgba. Pokémon Pinball on gbc let me set the strentgh of rumble but no rumble exists. 

[image]

lokisilverblade, 2021-01-17:
[image]

christianhaitian, 2021-01-18:
I had tried Pokemon Pinball on GBA.  I just tested Pokemon Pinball on GBC and it rumbles for me when I select the MGBA_Rumble core.  Did you make sure to select this core as it's not the default core for GB/GBC.

lokisilverblade, 2021-01-18:
I set mgba rumble for Pokémon Pinball on gbc and rumble work. But on gba the rumble option is still outgreyed.

christianhaitian, 2021-01-18:
What version of Pokemon Pinball are you trying for GBA?  The USA version of Pokemon Pinball rumbles fine for me.

lokisilverblade, 2021-01-18:
Its the european Version and the rumble works while installing 1.4 mid december. But Sound was stuttering and fps are bad.

lokisilverblade, 2021-01-18:
Ok the US rom have outgreyed rumble options too,

christianhaitian, 2021-01-18:
Have updated the new install yet?

christianhaitian, 2021-01-18:
I think I found your issue.  Go to the retroarch menu by hitting select+x while in game, then scroll down to options, hit A.  Then scroll to the bottom to Enable Game Boy Player Rumble and make sure it’s on.  If it wasn’t on, click a on it to turn it on, then hit b.  Then hit select+start twice to exit, then go back into the game and check to see if it rumbles now.

lokisilverblade, 2021-01-18:
Yes i have updated after Reinstall. And yes now the Rumble works but the Music stutters and the rumble is not exact. Some framedrops exists too.

christianhaitian, 2021-01-18:
I don't seem to have these issues on my end.  Make sure you don't have Frameskip enabled in the options menu while in game.

lokisilverblade, 2021-01-19:
Frameskip is disabled and everything is stock. I have changed only the rumble feature. You can close this here. My problem was solved and the other problems are only with pokemon pinball gba.

closed 2021-01-19 by christianhaitian