christianhaitian/arkos #202: Unable to Connect to WiFi

0o-R3MEDY-o0, 2021-01-17:
We have a mesh WiFi system in the house which ArkOS doesn’t seem to want to connect to, if I choose to connect to the mesh network and type in the password nothing shows up in the 'Connections' list. I have managed to add the main router to the 'Connections' list however when I go to activate the connection it doesn’t seem to work ?... I am using the ‘EDiMAX EW-7611ULB’ WiFi adapter which has been recommended to work with ArkOS and have also tried using multiple OTG adapters as well... I have managed to confirm that this is not a hardware issue as I can set up a hotspot from my phone and connect completely fine to that, it just doesn't seem to want to connect to the router. Any help would be appreciated. Thanks.

Device: RG351P

Version of ArkOS: v1.5

christianhaitian, 2021-01-17:
That unit is based on the rtl8xxx chipset which tends to be finicky in Linux.  You can try the fix posted in the known issues section of the wiki [here](https://github.com/christianhaitian/arkos/issues/124#issuecomment-751475195).  If that doesn't work, you can try the test version of ArkOS 1.5 [here](https://mega.nz/file/PcoBlAST#cN9oVCplJRu0nzqwtowP1-aJkJpEBnYRhKFBhni28M4) then the fix I linked to earlier and see if that improves the situation.  If not, I have no other suggestions other than to try a wireless adapter based on the mt7601u chipset.

closed 2021-01-17 by christianhaitian

0o-R3MEDY-o0, 2021-01-17:
Hi Christian, thanks for the reply. I have tried what you have suggested but unfortunately that hasn't worked. If need be I'll get it updated through the phones hotspot. Thanks for trying.