christianhaitian/arkos #203: 256GB Capacity.

nfabia90, 2021-01-17:
256 GB memory card is not compatible to arkos.?
And I am using a 256 gig PNY memory card.

christianhaitian, 2021-01-18:
I'm not sure as the max I've tested is 128GB but others have said that they've used 512GB without issue.  Did you follow the instructions on the main page for imaging and use recommended software for imaging?

microton79, 2021-01-18:
256 GB working fine for me

Wakaka1115, 2021-01-18:
Works fine for me too

nfabia90, 2021-01-18:
But it takes to long for me.

nfabia90, 2021-01-18:
> I'm not sure as the max I've tested is 128GB but others have said that they've used 512GB without issue. Did you follow the instructions on the main page for imaging and use recommended software for imaging?

Yes I used windisk32.

nfabia90, 2021-01-18:
But the device it self can only do 256 GB capacity.

nfabia90, 2021-01-18:
I forgot to mention when it loads it freezes on me when it loads on partition altered..

nfabia90, 2021-01-18:
Including I don't know if it requires fat32 or exfat format?

christianhaitian, 2021-01-18:
It takes care of setting the format and resizing on first boot as long as their isn’t an issue with the card and it’s of good quality. I recommend authentic Sandisk and Samsung cards

nfabia90, 2021-01-18:
> It takes care of setting the format and resizing on first boot as long as their isn’t an issue with the card and it’s of good quality. I recommend authentic Sandisk and Samsung cards

When it resizing the lettering and the numbers are in yellow not white.

christianhaitian, 2021-01-19:
Ok? Does the process ever complete?

nfabia90, 2021-01-19:
> Ok? Does the process ever complete?
No it just freezes when it's making its petition.

christianhaitian, 2021-01-19:
What is the make of the sd card?

nfabia90, 2021-01-19:
> What is the make of the sd card?

PNY 256 GB.

Wakaka1115, 2021-01-19:
I'm using Sandisk Extreme 256 gb. No issue at my end. Everything works just fine.

alpaga77, 2021-01-20:
i use Lexar 512 GB from Aliexpress - works fine :)

nfabia90, 2021-01-20:
256 GB Samsung Evo select works.
But I am not sure sure about PNY memory cards?

christianhaitian, 2021-01-20:
I've had no issues with the few PNY cards I've tried in the past but apparently the one you have doesn't work with ArkOS.  Could be the card is bad or simply incompatible with ArkOS for some reason.

nfabia90, 2021-01-21:
> I've had no issues with the few PNY cards I've tried in the past but apparently the one you have doesn't work with ArkOS.  Could be the card is bad or simply incompatible with ArkOS for some reason.

The memory card works it just does not work with the firmware for some reason and when I redid the whole entire process a couple of times it did the same thing with yellow lettering and numbers but it work with a Samsung card that I have just recently bought that is a 256 gig.

christianhaitian, 2021-01-21:
Ok.  Enjoy

closed 2021-01-21 by christianhaitian