christianhaitian/arkos #205: Blue screen on game load and close?

SRGilbert, 2021-01-17:
Not a big deal, but today after the recent update I’m getting a bright blue screen with a curser in the bottom left when I load a game.  It does a similar thing when closing the game.  I disabled the splash screen a while back, could that be connected?

christianhaitian, 2021-01-18:
There's nothing in the recent update that would cause that.  Can you post a screenshot of this blue screen?

SRGilbert, 2021-01-18:
[image]

SRGilbert, 2021-01-18:
Here you go 

[image]

SRGilbert, 2021-01-18:
Like I said, not a huge deal.  The only other change I made was adding Anberports.

SRGilbert, 2021-01-18:
Sorry, never mind. I just did a full power cycle and it went away.  🤷🏼‍♂️

closed 2021-01-18 by SRGilbert