christianhaitian/arkos #210: No controls/global hotkeys with WDLoad Amiga ROMs that default to mouse in Amiberry

fry-guy, 2021-01-19:
Amiga ROMs that default to mouse as their primary input method in Amiberry due to the configuration settings in the WHDLoad database result in a complete lack of controller compatibly -- as well as the inability to use global hotkeys to exit back out of the game. The only option is to reset the system.

I’ve determined that the mouse default is the problem since you can solve the hotkey issue by altering the individual game configurations in whdload_db.config (located under opt/amiberry/whdboot/game-data/) to switch PRIMARY_CONTROL from MOUSE to JOYSTICK and PORT0 from MOUSE to JOY. This does not solve the controller issue - you still can't play the games that require a mouse -- but it does allow you to use the hotkeys to escape the emulator. Presumably, the mouse setting blocks your ability to use start + select hotkeys to exit since it causes the system to look for input from a non-existent mouse.

As far as I am aware, this means there is no way to play games whose primary input method is mouse based. So I would regard this submission as a combined bug report/feature request.

Bug - Mouse-based games deactivate joypad controls and require a hard reset to exit the game when using WHDLoad games in Amiberry.

Feature request - Map an emulated mouse to the left joystick in Amiberry and default to this setting rather than a real mouse, which obviously exist on a handheld device.

This report is for ArkOS version 1.5 on the RG351P.

christianhaitian, 2021-01-20:
What are some example Amiga games that I can use to recreate this issue?

fry-guy, 2021-01-21:
Barbarian I and II are a good example. Barbarian I defaults to mouse as the primary control, and you encounter the problem with no controls or hotkeys, while Barbarian II defaults to joystick as the primary control, and plays fine.

I also encounter the mouse problem with Black Crypt and Might & Magic II: Gates to Another World. I can test some more later if you need additional examples.

The game files preconfigured with WHDLoad are findable at http://www.whdownload.com/.

Thanks for looking into this!

christianhaitian, 2021-02-08:
I haven't been able to find a good global solution for supporting both mouse and joystick.  The best solution I've found so far is to make these port changes on a per game basis and save the configuration for that game so it can load the configuration for that game.  I've added that information to the FAQ under the global events section for Amiberry.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS

closed 2021-02-08 by christianhaitian

fry-guy, 2021-02-14:
Thanks for looking into this. However, I think I'm missing something? The new documentation says to hit start + X to enter the Amiberry menu and switch to mouse controls on a per game basis. However, when a game defaults to mouse on launch - due to the instructions in default WHDLoad database file - it bypasses the jot as buttons. That is, you *can't* use any of the joypad keys - at all. So (as far as I am aware) you can't enter this menu to change the settings as the documentation recommends.

christianhaitian, 2021-02-14:
When I tested the games you suggested (Barbarian 1 and Barbarian II), I did not experience this issue.  I was able to get to the Amiberry Menu using the Select+X feature without issue.  So I have not been able to replicate this lack of access to the Amiberry menu that you state.  Were there any changes you made to Amiberry that may have possibly caused this?  If not, can you try imaging another SD card with ArkOS, update it only, and try the games you mention on that and see if the problem persist there are report back your findings?