christianhaitian/arkos #213: Daphne emulator no longer working after recent update

yeahguy72, 2021-01-22:


christianhaitian, 2021-01-22:
Still works on my end on 2 tests on my end with Dragon’s Lair 2.1.  Can you provide more information on your issue?

yeahguy72, 2021-01-22:
It’s the rg351p, I select the game and shows arkos log and closes right away and goes back to game list.

yeahguy72, 2021-01-22:
I had dragon’s lair 1 amd 2 working, space ace and cliff hanger. But now none work.  My console did have issue, so I reloaded arkos and the roms, bios and ports.

yeahguy72, 2021-01-22:
Is it possible to have a micro SD card become faulty and have to use a different one?

christianhaitian, 2021-01-22:
Did you make sure to follow the instructions [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#daphne) closely in reloading the daphne games here?

yeahguy72, 2021-01-22:
Yes, I did it and it worked until I updated it.

yeahguy72, 2021-01-22:
Updated the arkos software.

yeahguy72, 2021-01-22:
When the emulator failed, it freaked out the card and the system would no longer boot up, so I had to un-allocate and reformat the card, then reload arkos, update the os and then reload bios, roms and ports.

yeahguy72, 2021-01-22:
So I'm wondering if I should use a new micro SD card rather that the one that failed, even though I completely erased and reformatted.

christianhaitian, 2021-01-23:
What is the make of your sd card?

yeahguy72, 2021-01-23:
A legit Samsung 128gb.

christianhaitian, 2021-01-23:
Unless it's faulty, it should be fine.  Do you have another sd card you can use to test?

yeahguy72, 2021-01-23:
I'm gonna get one and try it.

christianhaitian, 2021-01-24:
Any update on this?

yeahguy72, 2021-01-25:
Just got the new card and loaded ArkOS. Should know shortly.

yeahguy72, 2021-01-25:
Found my first arkos.img file that worked and I reloaded onto a new micro SD and daphne works now. May have had a corrupted file on my second download of the ArkOS.img.

yeahguy72, 2021-01-25:
What other laser disk games work? I have space ace, dragon's lair 1 and 2 and cliff hanger.

christianhaitian, 2021-01-25:
You’ll have to go through and try some more.  Thanks

closed 2021-01-25 by christianhaitian