christianhaitian/arkos #214: Option to Disable Boot Rumble/Buzz/Vibration

bwiggs, 2021-01-23:
When I first turn on my rg351p, it buzzes/rumbles/vibrates, which is kind of loud when I want to play late at night. Is there any way to disable that?

christianhaitian, 2021-01-23:
I wish. It's currently not possible unless Anbernic provides a solution for this which I'm not holding my breath for. :(

closed 2021-01-23 by christianhaitian

0xLinkz, 2022-08-25:
Can this issue be reopened? Since AmberElec doesn't have the startup rumble there must be a solution for this.

christianhaitian, 2022-08-25:
There is a solution available, I just haven't implemented this for the 351v for ArkOS as of yet.  I'm trying to do this in a way folks can choose to have this on or off.  I will not implement this for the 351p or m as I no longer support those 2 devices.