christianhaitian/arkos #215: Feature: L2 or R2 to skip next track for bgmusic

junior55104, 2021-01-24:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-01-24:
I'm not sure this will be something I will have time to review for implementation but I will leave it here for consideration for now.

christianhaitian, 2021-03-20:
After further review and consideration, this will not be something I add at this time.  I may reconsider this in the distant future.  Thanks,

closed 2021-03-20 by christianhaitian