christianhaitian/arkos #216: Is it possible to make available to download through RetroArch the (Amstrad CPC) CrocoDS core apart from the Caprice32 which is the only one available?

JCORR76, 2021-01-25:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-01-25:
Does it perform better than Caprice32?

JCORR76, 2021-01-25:
Well the interface of the emulator is way better and easier than Caprice32. I use it on the RG350 and it just runs Amstrad games so smooth cause its Onscreen keyboard is much clearer than the Caprice32 one which is completely ineligible due to its small size. Pressing Select you open the emulator options and L1 to open the onscreen keyboard.  You can assign Amstrad keyboard key to any of the console buttons so you can start Amstrad games easily which I find terribly difficult on Caprice32...

christianhaitian, 2021-01-25:
Ok.  I'm compiling it now and will have a backup for it that I'd like you to test before I include it in the next update.

JCORR76, 2021-01-25:
If it works the same as in the RG350 it will be perfect. Thanks!

christianhaitian, 2021-01-25:
I think you're on to something with this emulator for cpc.  It seems to work better than caprice32 just from the few minutes I spent with it and Prince of Persia.  Please follow the instructions below to test this and provide feedback on this when you can including any suggested key remaps if any.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test the new CrocoDS Amstrad CPC Emulator.  I've made it the default emulator for Amstrad CPC already with this backup.  Cap32 is still available and you can switch back to it following the instructions [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-how-do-i-change-emulator-cores-in-ArkOS)

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5870097/arkosbackup.tar.gz)

JCORR76, 2021-01-26:
I'll test it as soon as I get back from work! Then I have to create a folder called "backup" in the roms folder, right?

christianhaitian, 2021-01-26:
Per step 1 in the directions, If the backup folder doesn't exist already in the roms directory, yes do create it.  Thanks for testing when you can.

JCORR76, 2021-01-26:
Hi,  following your instructions on ES, I press Start then click on Advanced Settings (I see no "Options") but then again I cannot see the "Restore Settings" option you say... It's EmulationStation v2.10.2.2 FCAMOD Build 20210123200317... right?

christianhaitian, 2021-01-26:
It's not in the start menu.  You have to go to the options section as it shows like a game system.  It should be just past scummvm

JCORR76, 2021-01-26:
Ok, I'm testing it right now. The onscreen keyboard is not QWERTY but AZERTY and the file of numbers seems to not be working.

christianhaitian, 2021-01-26:
The keyboard shows as QWERTY for me.  What game were you playing that the keyboard was AZERTY?
[image]

JCORR76, 2021-01-26:
Yup the onscreen keyboard looks QWERTY but if you type anything on the clear Amstrad screen, they keyboard seems to be AZERTY. Just type QWERTY and it will show AZERTY

JCORR76, 2021-01-26:
If it helps in anyway, the emulator for the RG350 works perfect.

christianhaitian, 2021-01-26:
It doesn't as that unit is mip based so the compilation of it is different.

christianhaitian, 2021-01-26:
I found the issue.  The keyboard was set to French in the code.  Once I changed it, the qwerty keys seems to be working correctly but I didn't thoroughly test it.  Please restore and test this new backup and let me know what you get.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5874953/arkosbackup.tar.gz)

christianhaitian, 2021-01-26:
I found the issue. The keyboard was set to French in the code. Once I changed it, the qwerty keys seems to be working correctly but I didn't thoroughly test it. Please restore and test this new backup and let me know what you get.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5874962/arkosbackup.tar.gz)

JCORR76, 2021-01-26:
Yup now all the keys, numbers included seem to work ok! It's curious cause this version of CrocoDS (v2020-02-04) is more updated than the RG350 one (v2020-01-15). Another thing, but I think it's a problem with the RG351 screen is that pressing L1 for different resolutions, the Auto and 320x200 seem to be the same while in the RG350 there was a difference.

christianhaitian, 2021-01-26:
Yea it must be something with the RG351 screen.  I only see a difference between 320x200 and overscan.  I'll check the keyboard code again. I probably need to find that right code for it.

christianhaitian, 2021-01-26:
Try this one.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5875060/arkosbackup.tar.gz)

JCORR76, 2021-01-26:
Yup now the onscreen keyboard shows all the correct characters.

JCORR76, 2021-01-26:
I have a question: Now, can you use the Load and Save states from RetroArch apart from the ones from the emulator itself as well?

christianhaitian, 2021-01-26:
I don't know.  Test it and try it.  The libretro doc says savestates are supported but don't know if it's in game or core itself.

JCORR76, 2021-01-26:
Thanks so much for doing this possible. Now it's really easy to play Amstrad CPC games on the RG351!

christianhaitian, 2021-01-26:
No problem.  Let me know if you find any other settings that should be adjusted during your testing/playing.  I plan to include this in the next update later this week.

JCORR76, 2021-01-26:
Great, so now the CrocoDS core will be available to download in the Core Downloader section in RetroArch as well? I would change the Caprice32 core for this one, this is sooo much better, right?

christianhaitian, 2021-01-26:
Yes.  I've requested that the repo include this core and in the next update it will be made as the default emulator as it better than caprice32 in my opinion.  I've found some interesting cpc games to play because the emulator works better.

JCORR76, 2021-01-26:
Yup what I most like about this emulator is how easy you can redefine the keys to play those games that need some extra keys to play well. Well on another note, I think the emulator is a bit more polished on the RG350 than in the RG351 hehe, look at the menu to redefine the keys hehe:
[image]

christianhaitian, 2021-01-26:
hmmm, it looks better on the rg351p to me because the menu is better centered.

JCORR76, 2021-01-26:
Haha the centered menu is in fact from the RG350 version while in the RG351 is the one not centered.

christianhaitian, 2021-01-26:
Hmm....it's centered on my rg351 unit.

JCORR76, 2021-01-26:
How? You mean you see the Redefine keys menu as in the console which is up in the picture? I see that menu as in the console down LOL, why is that?

christianhaitian, 2021-01-26:
Here's my screen.  
[image]

JCORR76, 2021-01-26:
Ah that's ok, it looks the same as mine. The thing is that I think the menu in the RG350 is more centered right? Here's the 350 menu:
[image]

And one more question sorry: Is it now safe to delete the "\roms\backup\" folder and its content? Thanks again!

christianhaitian, 2021-01-26:
No the 351 menu seems more centered to me but it's ok as long as the emulator works well and it seems better than caprice32.  Thanks for bringing this wonderful emulator to my attention and for all your help in testing it.  Yes, you can delete the backup folder and it's contents if you want to.

christianhaitian, 2021-01-29:
This has been officially accommodated with this evening's update.  Just update from the options section.

closed 2021-01-29 by christianhaitian