christianhaitian/arkos #217: Games not playing with right core unless started directly from Retroarch

Annie131, 2021-01-26:
I am unable to load many of the games I'm playing as they need gpSP core which I cannot select from Emulationstation. I have to open Retroarch and start the games there or they won't run.

### What device are you using?
RG351P

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1.5

### Anything else you'd like to include that may help to review this issue or feature request?
Games I'm having trouble with are ROM hacks.

christianhaitian, 2021-01-26:
Feedback I read is that gpsp has very bad performance so that's why it wasn't added.  I will add it as a selectable core in Emulationstation in the next update.

christianhaitian, 2021-01-29:
This has been accommodated with this evening's update.  Just update from the options section.

closed 2021-01-29 by christianhaitian

Annie131, 2021-01-29:
Thsnk you!

Annie131, 2021-01-29:
Unfortunately, it still didn’t fix the issue. If I use the gpSP core after starting Retroarch,  the games run but if I set to gpSP through EmulationStation thaey stop at a white screen - not sure what parameter is different when gpSP launched through Retroarch rather than from EmulatioStation.but at least I can use my saves on the ones that don’go to a white screen.

Some of the one’s I’m getting the white screen issue with are Pokemon Snakewood and Pokemon Theta Emerald which work fine if I launch Retroanrch first then use gpSP core.

christianhaitian, 2021-01-29:
Do you have a gba_bios.bin file in your bios folder?

Annie131, 2021-01-29:
I see a _gba_bios.bin file and gba_bios.bin  listed.