christianhaitian/arkos #219: Doom port

fabiocattaneo, 2021-01-27:
I've followed the FAQ section and copied the doom wad file to the roms/doom directory.
If I try to run the game from the doom section in the main menu, my rg351p goes back to the menu screen.
On the other hand, if I run the game from the ports section (where doom is supposed to be deprecated soon), it runs just fine

christianhaitian, 2021-01-27:
Have you updated ArkOS to the last 1/24 update?  Also did you change the default doom emulator?

fabiocattaneo, 2021-01-27:
Yes, latest update. Tried both Retroarch and the standalone options. Today I reinstalled ArkOS from scratch, applied all the updates, redownloaded the doom wad file. Same. It works under ports, it doesn’t work under doom.

christianhaitian, 2021-01-27:
It sounds like it could be a problem with the doom.wad file you're using doesn't work with lzdoom.  For the ports section, it probably worked because the prboom.wad file is already included there.  Copy the prboom.wad file to the /roms/doom folder and set it to retroarch and it will work there too.  https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#doom

fabiocattaneo, 2021-01-27:
Nope, doesn't work :(
I've deleted everything in the roms/doom folder, downloaded the doom.wad file from the wiki link, then copied in the same folder.
I've set the emulator to Retroarch, from the edit metadata section.
By the way, is it supposed to work with Retroarch? Because the "auto" setting defaults to standalone

christianhaitian, 2021-01-27:
standalone is lzdoom emulator.  retroarch is the prboom emulator which is the same used in the ports section.  Did you make sure to have both the prboom.wad and doom.wad in the folder?

fabiocattaneo, 2021-01-27:
OK, I'm an idiot... I've copied Doom1.wad in the ports/doom directory but not in the roms/doom one... Sorry for taking your time.
Anyway, out of curiosity, now I see 2 entries in the Doom menu: Doom and prboom. Is it OK? Moreover, should I keep the doom wad file in the ports/doom folder as well?

christianhaitian, 2021-01-27:
It's ok that they both show.  They will because of prboom needing prboom.wad.  If you don't plan to play prboom, you can delete it.  You can keep the doom wad in the ports doom folder but the next image release (which will be 1.6) will not have the doom ports section anymore.  Enjoy!

closed 2021-01-27 by christianhaitian

fabiocattaneo, 2021-01-27:
Sorry to bother you again, I have 2 more questions: if I want to play prboom should i copy the doom file in /usr/local/share/games/doom/ of the root partition as explained in the prboom website? what about ultimate doom? it appears in the ports list but I can't find any instructions in the wiki page

christianhaitian, 2021-01-27:
no.  Just copy the doom file to /roms/doom as explained in my wiki.  Same for ultimate doom.

fabiocattaneo, 2021-01-27:
OK thanks, I will try